CountDeMonet/ArduinoProtonPack #1: Additional LEDs

SmilingJack85, 2018-04-05:
Hi there, 
in the wiring diagram (without the bargraph), there are four LEDs after the jewels of the cyclotron, but i can´t find any information on what those are doing?
Also...is it possible to just use four single LEDs instead of the jewels for the cyclotron? I can´t figure out how i would have to change the code for that.

acfq48, 2018-04-05:
Those are the vent lights.  They turn bright white when the fire button is
released after overheating.  The answer to your second question is no.  The
jewels are individually addressed in order.  If you replace the four jewel
lights with four separate neopixels, they will all turn on at the same time
and the turn off for about 3 seconds.  The reason for this is that the code
references the lights in a string.  I'm not looking at the code but for
example, 1-7 light first, then 8-14, the 15-21, and lastly 22-28.  Then the
cycle repeats itself.  You would have to rewrite the code.  Check out my
version of this code on YouTube.  Af33 is the username.  The code is also
on github.  I just blew a 12v lithium battery so I think I'll do a wiring
video in a few weeks when I replace it.  Upload yours when youre done.
Keep the community rolling!

SmilingJack85, 2018-04-05:
Thanks for the super fast reply!
Of course, i forgot the vent lights...that makes sense.
That was the answer i expected, but i hoped for something else :-)
So i guess i have to order some more jewels (why do they have to be so expensive).
Or i figure something out to use some 2812 LEDs from a strip.
I´ll keep you updated, when i got it working.

closed 2018-04-05 by SmilingJack85

grogking, 2018-04-05:
Im working on fixing the code to use single neopixels. Ill let you know when i get it working

Raktageno, 2018-04-05:
Do you have a diagram of the entire rig wired up with the custom boards? I've hit a wall, but I'm not sure if it's a failure to get the code to properly load or not.

SmilingJack85, 2018-04-05:
For the electronics part i spend about 200 (euro) when i order four more jeweils.
The pack will be build in a budget for now...maybe i'll upgrade later.
I don't worry about the hardware part, got some experience with electronics, but i don't know mich about the coding stuff.

acfq48, 2018-04-05:
Unfortunately I don't have a diagram.

Here's the link to my completed pack based heavily on count demonets code.
It runs off of an arduino mega with an LCD screen.  The code is in the
comments.

I've also bought the following on amazon for easy soldering:

Gikfun Prototype Shield DIY KIT For Arduino UNO R3 Mega 328P (Pack of 3
Sets) Ek1038x3

Adafruit Perma-Proto Half-sized Breadboard PCB - 3 Pack! [ADA571

Stereo 20W Class D Audio Amplifier - MAX9744
You need 12v to get the power up for this amp, but damn it's worth it!

Good luck!!!

CountDeMonet, 2018-04-27:
Hey sorry I just saw this. The spirit proton pack version of this code uses single neopixels for the cyclotron lights. 

https://github.com/CountDeMonet/ArduinoProtonPack/blob/master/Arduino/SpiritMinimial/SpiritMinimial.ino

It shows how I do single neopixels. Basically the entire chain from the first powercell to the last vent light is indexed. I'll admit I didn't do it the same way for all of them so that is confusing. In this spirit pack code I just defined the start and end addresses for the individual lights like this

`
int c1Start = 0;
int c1End = 0;
int c2Start = 1;
int c2End = 1;
int c3Start = 2;
int c3End = 2;
int c4Start = 3;
int c4End = 3;
`

In this code example I moved the cyclotron leds to another pin to make it easier to understand. but the code is similar. 

In the pull code, to fix it so that it's using 2 neopixel sticks (16 neopixels) for the powercell, individual neopixels for the cyclotron, and individual for the venting you would just change the definitions like this

Note that because we are using the neopixels sticks there will always be 16 of them in the chain. in this case I'm starting at the second light and illuminating 2-15 for the sequence. This allows you to adjust the animation so that it fits inside of the available powercell window and not use leds outside the window. 

`
const int powercellLedCount = 14;                                        // total number of led's in the animation
const int powercellIndexOffset = 1;                                       // first led offset into the led chain for the animation

// now from 16 on we'll specify the individual cyclotron lights
const int c1Start = 16;
const int c1End = 16;
const int c2Start = 17;
const int c2End = 17;
const int c3Start = 18;
const int c3End = 18;
const int c4Start = 19;
const int c4End = 19;
const int ventStart = 20;
const int ventEnd = 23;
`

xblix, 2019-12-26:
in the other way  like adding a led
i wanna add 1 button in this code
this button is for fire_button2 for tips of wand. all work well. but i wanna a use this button2 for switching between 8 mode too.
i wanna 
if safety_switch == 0  && fire_button2 == 1  
switch mode ++1

each mode maybe is in a switch case and represent change for sounds and lights color
animation/speed can stay the same for all mode

‐-----

I try different way for implemented a button2 who's switch mode wjen safety_switch == 0.
but no succes

entire code in the loop section need to be replicate in sort of "main' switch case and where void animated lights function be replicated with minor change for color example : slime mode = cyclotron green light color & noseJewel  green light color

what could be the best way to implement this switch mode function and button2 without changing all the original code?  thanks if you could help a bit

xblix, 2020-01-14:
> in the other way like adding a led
> i wanna add 1 button in this code
> this button is for fire_button2 for tips of wand. all work well. but i wanna a use this button2 for switching between 8 mode too.
> i wanna
> if safety_switch == 0 && fire_button2 == 1
> switch mode ++1
> 
> each mode maybe is in a switch case and represent change for sounds and lights color
> animation/speed can stay the same for all mode
> 
> ‐-----
> 
> I try different way for implemented a button2 who's switch mode when safety_switch == 0.
> but no succes
> 
> entire code in the loop section need to be replicate in sort of "main' switch case and where void animated lights function be replicated with minor change for color example : slime mode = cyclotron green light color & noseJewel green light color
> 
> what could be the best way to implement this switch mode function and button2 without changing all the original code? thanks if you could help a bit

Done by StateMachine.
I add fire_button2
i add quick venting fuction
i add mode switch (13 different mode)

All by stateMachine function.

Thanks

grogking, 2020-01-14:
I was able to do this, but I had to make several changes in the code. I basically used the tip button to change the wand firing sounds and colors, as well as the colors of the cyclotron lights. Is that what you are trying to do?

xblix, 2020-01-14:
Yes and i do it is done!

I add a tip button for button_fire2
i could use this new button for a second firing sequence (example streamblast fire_button and  bosondart fire_button2. Perhap both could firing in the same times cause i use wav trigger board instead sfx.