CountDeMonet/ArduinoProtonPack #2: Can´t get the soundboard working

SmilingJack85, 2018-05-09:
Hi again, 

@CountDeMonet 
Sorry for opening another issue...and of course, thanks for your help!
I started building the full thing, with all the jewels :)

But right now i´m pretty desperate.
I can´t get the arduino to communicate with the soundboard i think.
With help i got the light working when changing the code so it works without the soundboard.
Also the soundboard itself works fine when testing, but i can´t get them to work together and i have no idea why.
I´m sure everything is wired correctly. 
I´m using the FX Soundboard mini (https://www.adafruit.com/product/2341), but that shouldn´t make any difference right?

Would really appreciate any kind of tipp or advice.

acfq48, 2018-05-09:
Can you send a few pictures of your wiring?

SmilingJack85, 2018-05-10:
@acfq48 ...man you're fast, thanks!
Gonna try later, took everything apart again and have to sleep a bit, its 4am Here :-D

SmilingJack85, 2018-05-10:
https://photos.app.goo.gl/Fi0BcPEoPq1TCgO83

Hope you can See something on the pics. I know its a Messer, but its Just for Testing.

What i'm wondering about is the 5V of the arduino...isn't it supposed to power it via VIN instead of giving 5V out to the breadboard, which is already powered?

acfq48, 2018-05-10:
Its possible that the board isn't getting enough power.  Here's a snippet I
found from the internet.

I believe I can supply the Arduino with power in three different ways:

   - Use USB. The design trusts the 5V to be regulated, and does not
   regulate it in any way.
   - Use a regulated power source, trusted to be 5V, and connect it to GND
   and 5V.
   - Connect an unregulated power source, eg a battery, to GND and VIN.
   This should be above 6.2V (since the dropout voltage is 1.2V) and
   preferably between 9V and 12V. There is a built-in regulator that will
   supply the Arduino with exactly 5V.

When I built mine, I used an arduino mega so your results may differ.  But
if I had a problem with the board, the best idea is to take the separate
parts and test them one by one to troubleshoot for the problem.  You might
be able to do what I did and hook up a usb connector to the arduino and
plug it into a laptop or small 5v 1 or 2 amp power pack and see if the
light on the arduino board comes on.  If I remember correctly, I think I
might have slowly turned the power regulated chip up from like 4.5v to like
6 v and everything just sprung to life.  But don't do that unless your
willing to get new gear for whatever is connected to the board.  (I blew an
LCD screen doing that.)  https://youtu.be/Lqbb-O-3d8M

The beat initial suggestion that I have is to get all of your lights going
first.  If the lights aren't on and cycling with the button presses as they
should, you're shooting blind on the soundboard.  Of what I could see, your
wiring looked pretty good.

Best of luck and let us know how it's going!!!

SmilingJack85, 2018-05-10:
ITS ALIVE!!!

I don´t know exactly what was wrong. I had some small issues, but nothing in particular why it didn´t work at all. Maybe it was just the combination of the little problems.
Anyway....its working now :-)
https://youtu.be/ziGLi3It3Ig

Thanks for your help!!!

@acfq48 Your pack is frikkin awesome! Would love to build something like that someday...if i had the time and the knowledge.

closed 2018-05-10 by SmilingJack85

acfq48, 2018-05-10:
If you're looking to build my version, the code is here.  I use an arduino
mega but the set up is very similar.  I think you can even use an uno if
you wanted to.  They both just adds the pinouts for the LCD screen.

https://github.com/acfq48/Arduino-Proton-Pack-With-LCD-Screen

Keep us updated!

CountDeMonet, 2018-05-17:
Hey sorry I missed this. For some reason I'm not getting emails from github and I don't check it every day. Glad you got it resolved and it's working. Looks great.

SmilingJack85, 2018-05-18:
@CountDeMonet 
No problem, thanks anyways! Works like a charm yes... but  @acfq48 gave me another idea :-D

Maybe you guys can help me again (i already feel bad for bothering you with my "ignorance")
For the last hours i tried to combine your codes, to use more dialogue and music tracks.
The dialogues work, but i can´t get the music to work. It only plays the first track (Ray Parker Jr.).

Have to confess i don´t really understand the thing with the pointers, @acfq48 is using "const" and @CountDeMonet is using "char" for the tracks, and i guess somewhere here i just don´t get it.

Like i said before, i don´t know much about coding languages, so i would be really thankfull if you could help me with that...if i get the chance someday i would definately buy you guys a beer/coffee/lemonade :-)

CountDeMonet, 2018-05-18:
I figured out where I was missing a check and I'm now getting emails yay :) Not a problem asking questions. Happy to help.

The difference you are seeing in the code is just a memory optimization. There was a memory issue found when using the string constants to hold the track names. I'm guessing acfq48 pulled the code he based his on before the optimization. The strings were causing us to only be able to use so many tracks with the adafruit board. The solution to that issue was to switch to a char array which uses much less memory. On my new project I have over 50 tracks in use and will probably end up with over a hundred when I'm done and I'm not having any issues with memory. 

The const part tells the compiler that this variable will never change and is a constant value. It's a compiler optimization for reduced memory usage.

acfq48, 2018-05-18:
Yeah...  what he said.  😀

I based my code off of demonets so listen to him.  Also, I used an arduino
mega so I guess I didn't care about memory optimization.  That and I'm a
noob who can barely write code so it's not very elegant.

Count, what kinda tricks are you working on over there?  Care to share with
the crew?

CountDeMonet, 2018-05-18:
Due to the fact that I've got so many tracks in the next project I'm not creating globals like I did for the proton pack. I have about 20 tracks that I really care about knowing so I reference those by name when I call the play function and the rest of it I reference by track number in a random way. It's an astromech droid like R2D2 so there are a lot of sounds that are random so the more the better. All of the code is posted here:

https://github.com/CountDeMonet/ArduinoAstromechDomeController

current status video. 
https://www.youtube.com/watch?v=Q4jPT27Jjtw

I'm actually installing the speaker amp, audiofx board, and speakers tonight to get him talking :)

SmilingJack85, 2018-05-18:
This is the droid i´m looking for! Thats amazing!
What is the little guy made of? Looks a bit like you 3D printed the parts?

Thanks for the explanation!
Got it working now and can´t stop smiling :-D

CountDeMonet, 2018-05-18:
Glad you got it working. The mega has tons of memory compared to the nano or even uno so sometimes tricks are needed and it's not always clear why. They don't fail in clear ways. 

The droid is mostly 3d printed with lots of hardware holding it together. Been working on it since late december. Hope to be doing the finishing work soon. Almost done finalizing the electronics and testing the drive train. It's pretty quick :)

SmilingJack85, 2018-05-18:
Its working, so better leave at it is now.
I don't know If i would have to change something If i'd switch to a mega, so thats no option for me :smile:
Maybe i'll try...got one laying around here.

Really great work! 
Can't wait to see it finished.
(I told my wife we need a 3D printer, maybe that will finally convince her)

Thanks again for your help!
Now i can sleep in peace...its already 4am here again :joy:

CountDeMonet, 2018-05-18:
I was just mentioning the mega because it is a different platform than the other ones. It's a good option if you need a lot of inputs but if you're trying to keep things small and lightweight the Nano is my preferred.

I love my 3D printer. I printed a proton pack and now I'm almost done with it Astromech droid so there's a lot you can do with it. Plus my wife has me making all kinds of things for the house like mounts for this thing or that thing.

acfq48, 2018-05-18:
Yeah, I used the mega because I needed the sda and scl outputs for my LCD
screen.  That and the fact that I had no idea what I was doing and thought
that all arduino systems were interchangeable. Wrong...

It took me some time to fiddle with the code to get it to work on a mega.
Since it's all in the proton pack anyway I didn't much care for the space
savings.  I'm still trying to talk count into helping me put the GB the
video game sounds and neopixel color changes in i.e. slime mode etc. But
I'll have to be patient for this awesome droid to be completed.  😉

SmilingJack85, 2018-07-05:
Hi again!
I didn´t know where to post this, so i hope its ok to use this thread one more time :-)
We had to rush some things, to get everything done for the convention, but the packs did turn out pretty good:
https://youtu.be/xv4iKZd05xQ

CountDeMonet, 2018-07-05:
@SmilingJack85 Looks great! nice work.