CountDeMonet/ArduinoProtonPack #3: Vent light question

ryandcole, 2018-06-13:
This looks awesome.  What neopixels did you use for the venting lights on the pack and wand?

CountDeMonet, 2018-06-13:
Thanks. This is the set I bought. Have a few left over still :)

https://www.amazon.com/gp/product/B01AG8X1X8

ryandcole, 2018-06-13:
Awesome.  Thank you!  Starting to plan the build now with the Spirit pack and updated 3D printed wand.  Hope to have it ready for Halloween!  Thanks!

closed 2018-06-13 by ryandcole

CountDeMonet, 2018-06-13:
Good luck. I really enjoyed this build. Happy to answer any questions that come up

ryandcole, 2018-06-13:
Awesome.  I'm sure that I'll have them!  Below are the goggles I made for last Halloween.
[image]

acfq48, 2018-06-13:
What wire did you use count?  It's triple stranded and nicely kept
together.  Mine kinda looks like a hodgepodge of 24 gauge garbage...

CountDeMonet, 2018-06-13:
I used a few different kinds. For the wires that are connected I used this ribbon wire

Uxcell a16030200ux0389 Ribbon IDC Cable Wire Rainbow Cable Flat Color Rainbow, 2 m, 6.6', 40 Way, 40 Pin https://www.amazon.com/dp/B01JRA853W/ref=cm_sw_r_cp_taa_i_v3AiBbWW0KRZ4

I also used it for the pack ribbon wire that is displayed on the outside.

I like this stuff for pretty much everything else. Very flexible and strong 

Striveday&trade;Flexible Silicone Wire 24awg Electric wire 24 gauge Coper Hook Up Wire 300V Cables electronic stranded wire cable electrics DIY BOX-1 https://www.amazon.com/dp/B01LH1G2IE/ref=cm_sw_r_cp_taa_i_t6AiBbMJKFZTC