CountDeMonet/ArduinoProtonPack #4: Cant compile with bar graph

G1Soundwave, 2018-08-12:
Ive been fighting with this a bit now, the code without bar graph works great, but with bargraph, it stops compiling and gives this error- "'shutdown_leds' was not declared in this scope" I would try to fix it myself, but to be honest, I know almost nothing about coding. But I will say, even without the bargraph, excellent job on it!

CountDeMonet, 2018-08-12:
That is really odd. The code compiles for me. I would make sure you have pulled the latest code from the repository and retry compiling. That function it is complaining about is down at line 1102 and should look like this

`
void shutdown_leds() {
  // reset the sequence
  seq_1_current = 1;
  fireSequenceNum = 1;

  // shut all led's off
  for (int i = 1; i <= 15; i++) {
    switch_graph_led(i, LOW);
  }
}
`

acfq48, 2018-08-12:
For anyone interested:  Spirit Halloween is releasing a ghost trap and PKE
meter among a few other new Ghostbuster items.  They're running for $50 and
$35 respectively and look pretty good.  I'll be working on a foot pedal mod
for the trap as soon as I get it but if anyone beats me to it, be sure to
post.

https://www.spirithalloween.com/product/ghostbuster-ghost-trap-ghostbusters/164228.uts?keyword=ghost
trap&thumbnailIndex=1&Search=Find+It

Enjoy!

G1Soundwave, 2018-08-13:
Not sure what happened, but seems it was missing those sections in the code. Redownloaded it, and working fine. Dont know why I didnt think to try that. Thanks alot.

closed 2018-08-13 by G1Soundwave