CountDeMonet/ArduinoProtonPack #5: No booting up

fstanley28, 2019-01-23:
I have everything loaded on the soundboard and nano both power up(lights turn on on both boards) but I get nothing when I flip switches. I believe I have everything wired correctly. Any help would be appreciated.

acfq48, 2019-01-23:
Post some pictures or a video and link to it.

CountDeMonet, 2019-01-23:
A while back I removed the stop if there are issues with the sound board. It is the most finicky part of the whole thing. It should be working assuming all connections are good. Check solder joints first and foremost. Pictures of the setup would help as well.

fstanley28, 2019-01-23:
I appreciate the super quick repsonses. I will get some video/photos up later tonight. Thank you!

fstanley28, 2019-01-24:
Here is the video link: https://youtu.be/wZFvqCkVhzI

Here is a link to a few photos:
https://drive.google.com/drive/folders/1hw2k3sZqJJ9PCTk1Jc0FeTagvnJx77Eg

My wiring is a bit of a mess. The last time I wired up anything up, I was 18 working at RadioShack.

fstanley28, 2019-01-24:
No they have never swtich on. I have not tried it with out the soundboard. I wll try with out the board momentarily.

CountDeMonet, 2019-01-24:
I'd recommend loading up the spirit pack version wish is really minimal and only does  the lights. In this version the powercell and cyclotron lights are split up so you can only get the powercell working but it ignores switches and the sound board and you can see if the powercell works at a minimum. It's good to see something work when you are starting in on these projects. 

https://github.com/CountDeMonet/ArduinoProtonPack/blob/master/Arduino/SpiritMinimial/SpiritMinimial.ino

CountDeMonet, 2019-01-24:
I'm having an issue with google drive so I can't see the pics for some reason. Hope I'll be able to later. I watched the video and it is hard to make out how everything is hooked up. There are a lot of wires in this project :) Sometimes I find it's best to test components individually when there is a general problem like this. Maybe use some example switch and button sketches to test the input components.  I would definitely start with the switches since those are what control everything else. 

If you have a multimeter verify the voltages at various places around the circuit.

fstanley28, 2019-01-24:
@countdemonet I’ll give the spirit code a try. I uploaded the pictures in another format. The first set was in some weird iPhone format.

fstanley28, 2019-01-25:
@countdemonet got it working with the spirit only.  https://youtu.be/We_NuNtLlww

On to more testing.

acfq48, 2019-01-25:
Great job!  That at least proves that your initial setup and arduino work.
Count, do you recall if the code works for all of the blue lights or does
it skip one?  I suspect that one of the neopixels on this rig is burnt out
and the stick needs to be replaced.

CountDeMonet, 2019-01-25:
Yeah that is interesting. The default code does have a limit built in to show 14 leds in the sequence (how many fit in the spirit window). It looks like it was updated to show all 16. It is LED #7 in the chain if you wanted to try and change the color directly.  

this would try to make the LED bright red. Comment out the current loop and then place this in there. You could test all of the leds 1 by 1 in the chain with this.

`
void loop() {
  powerStick.setPixelColor(7, powerStick.Color(255, 0, 0));
  powerStick.show();
  delay(1);
}
`

So now that we know that the led's work for the most part the next bit to figure out is why is the full code not moving forward and running. Checking the switches is your next step.

fstanley28, 2019-01-26:
All lights are now working! No sound out of the sound board. Seems to not be triggering. Getting a small pop from the speakers when the battery is plugged in, good news there. Has anyone had defective Adafruit soundboards? I plugged it it and loaded it initially but now when I plug it into my comp it’s not recognized. Maybe it’s defective. I’m going to try a replacement m. Thanks for everyone’s help!

CountDeMonet, 2019-01-26:
There is a sketch installed with the adafruit libs that is great for testing the audio board. It let's you interactively control from the Arduino using serial monitor. I use this a lot to get everything setup on the fx board and make sure everything is working. Just update the PIN numbers in the example sxetch and upload it to Arduino.

https://learn.adafruit.com/adafruit-audio-fx-sound-board/serial-audio-control

fstanley28, 2019-02-08:
Just an update on progress. Got the sound working. Everything is firing as it should! Now to put everything together for the finished pack. I appreciate everyone’s help here.

acfq48, 2019-02-08:
Awesome news!  Post a vid when you're all wrapped up.

CountDeMonet, 2019-02-08:
Excellent! Yes please share a video I'd love to see it

closed 2019-04-22 by CountDeMonet