CountDeMonet/ArduinoProtonPack #6: help

BSKStudio, 2019-03-13:
This is something probably that I'm doing wrong.  Im having a issue when Im importing the zip file into my library?  its not allowing the queue array to and other files to import?

CountDeMonet, 2019-03-13:
Make sure you install the libraries to the library folder for the ide

https://www.arduino.cc/en/guide/libraries#toc3

BSKStudio, 2019-03-15:
Awesome thank you!

closed 2019-04-22 by CountDeMonet

Autocado, 2021-08-15:
how do I uploads the code and sound

CountDeMonet, 2021-08-15:
@Autocado you are going to want to review this page on getting started with the arduino

https://www.arduino.cc/en/main/howto

and this page on uploading files to the audiofx board

https://learn.adafruit.com/adafruit-audio-fx-sound-board/copying-audio-files