CountDeMonet/ArduinoProtonPack #7: Strange problem.

criticalkaos, 2019-04-22:
I have built all the circuit using a breadboard all worked fine, i bought a solderable breadboard and i am slowly moving everything over, all powers up but I am having a strange issue I have never come across when the board is laying flat the leds run slow when I put it on its side it speeds up to normal speed this also happens even if i wave my hand in front of the back of the board not touching it, I have re soldered entire new boards 3 times now changing the nano 3 times also swapping the led modules, and cables but it still does the same im totally lost.

i have not completely wired everything up yet nano is soldered cyclotron, power cell and vent lights all wired and the wand light is wired with one switch for testing am i missing something?

ever come across anything like this?

CountDeMonet, 2019-04-22:
Slowdown would indicate an issue with the FX board if that is what you are using. Could be a bad ground or weak solder joint on the tx/rx pins. Slowdown is almost always a communication problem with the audio board though. I'd use the adafruit test program to make sure it's working properly

https://learn.adafruit.com/adafruit-audio-fx-sound-board/serial-audio-control

Just update the pin numbers in the sketch to the ones used for the audio board and upload to the arduino. The test sketch is the example included with the library for the arduino.

criticalkaos, 2019-04-22:
Thanks for the quick response at this point I dont actually have the audio board connected that all seemed to work fine on the breadboard at this point all I have connected is the pack lights. I was originally thinking it might be bad soldering but I have done it 3 times now using new protobaords new nanos even purchased new solder and used different cables but I get same result. It could be bad ground or some sort of interference as it does speed up if I touch the pins on the board. Scratching me head on this one.

CountDeMonet, 2019-04-22:
Are you using the full code and don't have the soundboard attached you need to make a mod to the code to disable the soundboard. 

Comment out

In the header

SoftwareSerial ss = SoftwareSerial(SFX_TX, SFX_RX);
Adafruit_Soundboard sfx = Adafruit_Soundboard( &ss, NULL, SFX_RST);
 
In setup()

// softwareserial at 9600 baud for the audio board
ss.begin(9600);

// set act modes for the fx board
pinMode(ACT, INPUT);

The entire contents of playAudio() and playDialogTrack() functions

and then in the main loop() change

int playing = digitalRead(ACT);

to

int playing = 0;

If you are not using the full code then are there any modifications you have made to the script?

criticalkaos, 2019-04-22:
Ooooh so my problem could just be the fact I dont have the sound board connected. I have been running the full code no changes made but I like to test things along the way, do a bit test it do some more test it again just using one neopixel strip connected if it runs then carry on I may not have noticed the speed difference up until now, I have been doing this in my spare time over the past couple of weeks so may of not noticed until now and just assumed it should work without the audio board connected.

CountDeMonet, 2019-04-22:
yeah the full code needs the audio board. I recently remove the RST code that would completely stop all initialization if the board was not found and to free up a pin. Try disabling the board and I expect things will speed up

criticalkaos, 2019-04-22:
Success it works fine now I have sound board connected I honestly thought it wouldn't make any difference but so far no issues. Thanks for the help it works a treat now.

CountDeMonet, 2019-04-22:
Great, glad things are working now

closed 2019-04-22 by CountDeMonet

criticalkaos, 2019-10-24:
Hi sorry to be a pain just noticed you have updated the code to include ecig do you have any more info how to connect up the relay to the arduino I have have everything ready to go.

ericbanker, 2019-10-24:
The signal pin for the relay is 13. Then you also want to connect the ground for the relay to your common ground for the Arduino and other components. From that point on the Arduino side of the relay is done. If the relay is working backwards from how it should look in the code and there are a few places where you can reverse the relay function.

criticalkaos, 2019-10-24:
Nice one cheers mate I'll give it a try tonight.