CountDeMonet/ArduinoProtonPack #8: Battery options?

SmokeFountain, 2019-07-26:
Hello, I'm an absolute beginner at this and giving this a shot for my proton pack build. I have zero experience with lipo batteries and read a lot of stories of them being dangerous when not handled properly (Dangerous when overcharging or when they get under 3V) so I was wondering if there was an alternative for the lipo that I could use to power this project for a decent amount of time. Sorry if this is a really nooby question but I'm  pretty clueless and this is my first arduino project so I'm learning along the way.

acfq48, 2019-07-27:
Check out my video on this.  I had the same question years ago and as it
turns out, the 12v lipo can be replaced with 8AAs for similar results.  AA
batteries produce 1.5v so 8x1.5=12. When placed consecutively, one after
the other of course.  They sell holders like that on eBay and Amazon for
about 8 bucks or so.

My pack uses both Lipo and AA and honestly, it was probably overkill.  I
just wanted to be able to take it to a con and if my lipo ran out then I
could switch and keep the pack going forever.  Just be sure to put the
current through the buck converter to downsize the voltage to 5v.  If you
pour 12v straight into your electronics it will fry them.  Lots of smoke
and smoke is bad.  Get a cheap voltage reader and find a video on how to
use it.  Put the 12v into the converter and read the other end while
adjusting the small screw on the converter.  Once you hit 5v, then and only
then connect the arduino.  (I actually think the arduino may take up to 9v
but I'm not sure enough to advise that)  if you use the adafruit sound
board I think it should take the full 12v but check the pinout drawing to
be sure.  If so, you just tap a line off of the 8AA before it goes to the
buck converter and then onto the board.

Hope that helps!

ericbanker, 2019-07-27:
I've been using lipos for more than 15 years now and as long as you charge them with a good charger there won't be any issues. I have packs that I have over discharged and the good charger won't charge them.

That said there are many options for powering any device. I like lipos because of the weight to capacity.  I use 11.1v 2200 mAh packs for the proton pack and I can get a full day of usage out of it. 

The voltage is the other part of this. Arduino likes 5v and most everything in this design used that voltage or lower. You could go with a two cell lipo at 2000mAh as well. I had many 3 cell from my rc days which is why I used that. As long as you are over 5v and a decent capacity you will be fine

Now on to the converter. The Arduino 5v out is very limited so I chose to power everything from the buck converter instead of the Arduino 5v out. You will run into issues due to the current draw if you tried to power all off the Arduino 5v or USB. The amp for audio also needs a very stable voltage for good sound. You can test it all off the Arduino usb as long as you keep the volume very low. For real world use power separate

Hope that helps a bit

SmokeFountain, 2019-07-27:
Awesome, thanks so much for the responses😊. I have the buck converters so I’ll definately use them. I have one of those 8AA battery holders lying around so I’ll use that for now and when everything is set up and working as it should I’ll give the Lipo a try for the sake of it being rechargeable. Thanks again😊👍🏻.

acfq48, 2019-07-27:
Best of luck buddy!  Post some vids of the finished product.  😁😁😁

closed 2019-11-19 by CountDeMonet