CountDeMonet/ArduinoProtonPack #9: Code Issues

kbritton87, 2019-09-21:
Hi mate, hoping you can help me, getting all the libraries sorted but keep getting the below issue.

libraries\SparkFun_SX1509_Arduino_Library\SparkFunSX1509.cpp.o (symbol from plugin): In function `SX1509::SX1509()':

(.text+0x0): multiple definition of `REG_I_ON'

sketch\Arduino_Code.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\SparkFun_SX1509_Arduino_Library\SparkFunSX1509.cpp.o (symbol from plugin): In function `SX1509::SX1509()':

(.text+0x0): multiple definition of `REG_T_ON'

sketch\Arduino_Code.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\SparkFun_SX1509_Arduino_Library\SparkFunSX1509.cpp.o (symbol from plugin): In function `SX1509::SX1509()':

(.text+0x0): multiple definition of `REG_OFF'

sketch\Arduino_Code.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\SparkFun_SX1509_Arduino_Library\SparkFunSX1509.cpp.o (symbol from plugin): In function `SX1509::SX1509()':

(.text+0x0): multiple definition of `REG_T_RISE'

sketch\Arduino_Code.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\SparkFun_SX1509_Arduino_Library\SparkFunSX1509.cpp.o (symbol from plugin): In function `SX1509::SX1509()':

(.text+0x0): multiple definition of `REG_T_FALL'

sketch\Arduino_Code.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

Multiple libraries were found for "Wire.h"
 Used: C:\Program
Multiple libraries were found for "SparkFunSX1509.h"
 Used: C:\Users\KarlB\OneDrive\Documents\Arduino\libraries\SparkFun_SX1509_Arduino_Library
Multiple libraries were found for "QueueArray.h"
 Used: C:\Users\KarlB\OneDrive\Documents\Arduino\libraries\QueueArray
Multiple libraries were found for "Adafruit_Soundboard.h"
 Used: C:\Users\KarlB\OneDrive\Documents\Arduino\libraries\Adafruit_Soundboard
Multiple libraries were found for "Adafruit_NeoPixel.h"
 Used: C:\Users\KarlB\OneDrive\Documents\Arduino\libraries\Adafruit_NewPixel
Multiple libraries were found for "SoftwareSerial.h"
 Used: C:\Program
exit status 1
Error compiling for board Arduino Nano.

Thanks
Karl

acfq48, 2019-09-21:
I was having similar problems and sometimes you just need to spend some
time and clean up your libraries.  Stay tuned.  I'm currently coding in all
of the video game weapons.  I just need to finish up some final wav files
to fit everything onto the adafruit board.

kbritton87, 2019-09-21:
Cool sounds good. I am new to do this and decided to do this as my first project, maybe a bit optimsitc, but trying to learn as I go. I will have a look through it just keep saying the function has multiple definitions, so not really sure how to fix this.

acfq48, 2019-09-21:
It probably means that you have multiple files in your library that have
the exact same .h name.  I recommend copying everything out into a new file
and slowly putting everything back in.  Checking for duplicates along the
way.

Also, if you're brand new, this is a pretty complicated project.  First
thing to do is to get the libraries and code set up correctly.  If you
compile the base code with no errors next wire up all of your switches and
one or two of the easiest to connect lights.  Since the lights are
addressable neo pixels, if you don't get the first one right, everything in
line behind it won't work.  The switch that starts the pack is the most
important.  No other buttons with work without it.  Once you get a few
lights working, then build out all of the lights and lastly do the sound
board.  It's also helpful to solder up an adafruit jewel with the power,
ground and data lines and then go around and test all of the different data
out pins for the neo pixels.  This will help you know, which line should go
with which just incase the pinout diagram is unclear.

ericbanker, 2019-09-22:
It does sound like you have multiple libraries with the same names. One option is to change the directory for libraries in the Arduino ide and then start bringing the libs needed in one by one. This project does not need many. Starting fresh seems like a good try and if that doesn't work you can always set it back to what it was before.

I'd recommend starting with the spirit minimal setup if you are very new to all this. Once that is working move on to full code.

kbritton87, 2019-09-22:
Thanks Guys,

I will start over and update the library slowly. See how that does!

closed 2019-11-19 by CountDeMonet