CountDeMonet/ArduinoProtonPack #10: Switch not working

J-to-the-R, 2019-10-06:
I have hooked up everything and when wire by wire to make sure I wired everything correctly. No luck when a switch on the start up switch. 

I also started from the beginning and tested the simplified Spirit version with just the two pins. That one works from me. When I attempt the more advance one with the switches I can not seem to be able to turn on the lights after turning on the start up switch.

Any recommendations? Thank you.

acfq48, 2019-10-06:
It's rare, but did you try changing out arduino boards?  The key to getting
this project or really any project working is to get one light and one
switch up and going.  The best thing that I can say is to remove everything
except for the start up switch and the cyclotron lights.  Follow the pinout
guide and try to get those items working.  If that doesn't work for you
send us pictures and well figure it out.  Many pictures that are close up
shots to see the numbers on the board.

CountDeMonet, 2019-10-06:
acfq48 is right, start simply. Which version of the full code are you going for? If it's the one with the bar graph it won't get past the initialization without the sx1509 initializing properly. I think that's the only blocking component in the code base now. If you don't have that hooked up yet and you are using that code that could be the issue. 

Simplifying down to the base start switch and cyclotron lights is a great idea and a good thing to start building on.

acfq48, 2019-10-06:
Eric,

I'm almost finished with the pack upgrades!  I'll send it to you once it's
finished but it adds the three additional weapons from the video game.
You'll need to add one additional button to the wand but I've been able to
compress all of the sound files (44 in total) plus 5 songs down to 14mb.
Ive still got a few more tweeks to do, maybe increase the quality for a
sound file or two, but I definitely recommend the upgrade.  The LCD screen
on the pack is alot of fun too and worth trying out.  You also may need to
upgrade to the mega.  My code runs on a mega but I'm sure you can figure it
out.

Best,

Alex

J-to-the-R, 2019-10-06:
That was exactly the issue. I was attempting to use the bar graph one without the sx1509 connected. Thanks. I testing the one without the graph and it worked as expected. Thanks you. Awesome project. Thanks again for putting this all together.

closed 2019-10-06 by J-to-the-R