CountDeMonet/ArduinoProtonPack #11: Sound won't trigger it.

xblix, 2019-11-18:
@fstanley28    how you fix your sound problem finaly.

same situtation.

all wires look perfectly hook up to the right pin.

i can trigger by ground pin all  but via uart....no sound and no trigger

any clue?

Just an update on progress. Got the sound working. Everything is firing as it should! Now to put everything together for the finished pack. I appreciate everyone’s help here.

_Originally posted by @fstanley28 in https://github.com/CountDeMonet/ArduinoProtonPack/issues/5#issuecomment-461767544_

fstanley28, 2019-11-18:
I forget exactly what I did. I believe though that I double checked everything and reloaded the sound files onto the soundboard and that fixed it.

CountDeMonet, 2019-11-18:
The pack code does not log to serial out which is why it is not doing a Serial.begin(115200);

Since you have the menucommands sketch working that is great. What is not clear from your video is what the proton pack code is doing. Are you getting any lights for the cyclotron and powercell when you flip on the main switch (pin 8)?

xblix, 2019-11-18:
Yes all my lights and switchs work and react like original code suppose.

CountDeMonet, 2019-11-18:
using the menucommands sketch I want you to type the list command (L) and then copy and paste the results into this thread

xblix, 2019-11-18:
I forget than i convert all .WAV on .OGG for save space on soundboard. After changing all .wav in .ogg ib the code all perfectly work now. Thanks @CountDeMonet for point me out the files list trick.

One thing not work properly is the vent light. The vent light show up only 2 second instead to show up for the entire venting sound time...any clue to augmented times the vent light show up in the code ?

CountDeMonet, 2019-11-18:
that is odd. The venting lights are controlled by the isVenting boolean variable. That is only set to false on startup and when the booting animation is complete so you should see the lights on for the entire boot animation. Can you take a quick video of what is currently happening?

xblix, 2019-11-18:
its look like the "vent animation" stop adtern1nsecond instead during all "vent sound" time
https://youtu.be/t5wr8_8CiaI

CountDeMonet, 2019-11-19:
Well it's not actually doing the full reset animation and looks like it does one second of the reset then back to the standard. Can you post the code you are using? Looks like you had to make some mods to it for that config.

xblix, 2019-11-19:
yellow cycle suppose to show too but no yellow

https://pastebin.com/1qScEVrw

CountDeMonet, 2019-11-19:
ah, I think I figured out what is going on. You are using 3 neopixels for the powercell animation. If you look at how the animations works they are timed with the sounds using intervals based off the number of neopixels in the animation. In this case the boot animation as that is the one triggered on a "venting". In the original configuration there are 14 neopixels in the powercell. That is the one that is dictating how long to "vent" for. 

void powerSequenceBoot(unsigned long currentMillis) // function in question. 

In the main powercell boot animation I have 14 neopixels involved and an update interval of 60 seconds. The animation starts will all empty and then does a follow pattern stopping at the last lights until all lights in the powercell are on. At 60ms with 14 neopixels that's about 6 - 7 seconds which is how long the audio file is. Since you reduced the number of neopixels for the power cell you will need to change the delay to take into account the reduced number of leds. 

const unsigned long pwr_boot_interval = 60;       // interval at which to cycle lights (milliseconds). Adjust this if 

Just a guess but it's probably going to need to be closer to 1000 ms as with the 3 led's that leaves only 6 "frames" for the animation I believe. Could be off on that.

xblix, 2019-11-19:
i dont change the code since yesterday cause i see something...
https://youtu.be/fhZpvA5-lyI
animation its ajusted by the const number of led and works wee without soundboard connected
but with sound board connected..
 not...only 1 second

CountDeMonet, 2019-11-19:
When you are disconnecting the soundboard you are causing delays to occur due to failures to communicate with the sound board. These are the pauses each time it tries to play an audio track. If you actually disable the sound board within the code and don't attempt to play the tracks or access the play state then you will find those delays go away and the code acts the same way. The issue is the timing of the intervals and the number of led's in the animations.

xblix, 2019-11-19:
const unsigned long pwr_boot_interval = 680;

all slick now.
thanks for help and this great code @CountDeMonet