CountDeMonet/ArduinoProtonPack #12: Real random

xblix, 2020-01-14:
i play since several week with this code and all is so good.

But i realize than queuarray is not a real random sounds playlist. Yeah after many test its appears than the queuarray suppose to play random quote of the movie but he not did it. Its always the same order after a reset of the board.

How i can make a real random queu?

CountDeMonet, 2020-01-14:
I tried a number of things to try and randomize the sayings without much luck. The random() function on the arduino isn't as random as you would hope. I kept getting repeating quotes. The queue array was an attempt to fix that but that didn't work either but the dequeing worked great. If you want to give a go at randomizing it yourself look at this code where the dialogs are entered into the queue. If you can randomize the order placed into the queue that would do it.

`
void playDialogTrack( int playing ) {
  // if the queue is empty reseed it
  if ( dialogQueue.isEmpty() ) {
    for (int i = 1; i <= numDialog; i++) {
      dialogQueue.enqueue(i);
    }
  }

`

grogking, 2020-01-14:
One suggestion I have is using the randomSeed() funtion with the analogRead () function. Essentially, it reads an unused pin which will give it a random number from the "noise" on the unused pin to generate a seed number that then is used as a seed to start the randomizer with an actual random number. 

Here is code I use for generating random blinking patterns on leds. 
`//Declare and initialize LED pin variables
int LED_1 = 2;
int LED_2 = 3;
int LED_3 = 4;

//this variable will hold a random number generated by the random() function
long randomNumber;

//Set up - this is where you get things "set-up". It will only run once
void setup() {
  
  //setup serial communications through the USB
  Serial.begin(9600);
  
  //Let's print a start messgae to the serial monitor when a new sequence of random numbers starts
  Serial.println("Starting new Random Number Sequence");

  //set the LED pins as outputs
  pinMode(LED_1, OUTPUT);
  pinMode(LED_2, OUTPUT);
  pinMode(LED_3, OUTPUT);
      
  //Let's make it more random
  randomSeed(analogRead(A0));   
      
}//close setup

//The loop() runs over and over again
void loop() {
  
  //generate a random number
  randomNumber = random(2,5);
  
  //display the random number on the serial monitor
  Serial.print("The Random Number is = ");
  Serial.println(randomNumber);`

xblix, 2020-01-14:
Exactly what i find too  randomSeed.
Code alonge serual print perfectly.
But integrate to the proton code, print is not random.
Close from the solution.
Perhap how i put the random number from Seed and where i put inside this : 
void playDialogTrack( int playing ) {
  // if the queue is empty reseed it
  if ( dialogQueue.isEmpty() ) {
    for (int i = 1; i <= numDialog; i++) {
      dialogQueue.enqueue(i);
    }
  }

grogking, 2020-01-14:
I'm not a coding expert, self taught mostly, and its been a while since I edited this code for my pack.  However, perhaps you need to get rid of the queue, and replace it with something using something like this
long quoteNumber = 0:
void setup() {
randomSeed (analogRead (put empty pin number here);
}
void loop (){
quoteNumber = random (1,8)   // I forget how many quotes there are

}

Then have it play the quote that matches the number in quoteNumber.

xblix, 2020-01-14:
Finaly i get randomNumber "really" random inside proton code.

`  long randomNumber = 0:
void setup() {
randomSeed (analogRead (A4);

void playThemeTrack( int playing ) {
  // if the queue is empty reseed it
  if ( themeQueue.isEmpty() ) {
    for (int i = 1; i <= randomNumber; i++) {
      themeQueue.enqueue(i);
    }
  }
}
void loop (){
quoteNumber = random (1,8) // 7 quote  - Max not include

}

  `

Thanks for help guys!!!

closed 2020-01-14 by xblix