CountDeMonet/ArduinoProtonPack #13: E-cig addition

fstanley28, 2020-01-28:
Has anyone tried this yet? I see there is and ino file with the relay but as for wiring I didn't see anything. Any guidance would be awesome.

CountDeMonet, 2020-01-28:
A number of people have used that code. As for wiring it up there is not much to it. The relay is a remote controlled switch so on the control side there is a ground and signal. Those connect to the arduino ground and pin 13. Then you wire the ecig up to it's separate power source and stick the relay in between the power side.  Here's an example of how it can go

https://i0.wp.com/randomnerdtutorials.com/wp-content/uploads/2016/11/Schematics.png?w=700&ssl=1

from this site https://randomnerdtutorials.com/guide-for-relay-module-with-arduino/

fstanley28, 2020-01-28:
Perfect, thank you. I am loving this build!

closed 2020-01-28 by fstanley28