CountDeMonet/ArduinoProtonPack #14: useGameCyclotronEffect

xblix, 2020-02-25:
i try to create a second mode : slime mode.
For that i create bosonmode bool and slimemode bool for swtich true or false between both mode. Like this 

`bool bosonmode = true;
bool slimemode = false;`

The code i do work super good when gamecyclotroneffet is false.
But when is true its not work.
When i switch mode from boson to slime mode, all stop (all function, not just light sequence)

There is my code part where is super good in bosonmode but not in slimemode 

[slimemode.txt](https://github.com/CountDeMonet/ArduinoProtonPack/files/4251348/slimemode.txt)

Any cue what wrong? or how to achieve this for have a perfect same pathern but in green fade effect instead of red when its slimemode is true?

i try to create new function (reproduce all red function in green, create 5th and 6th case in the switch too but nothing work. Always stop work when i switch.

I know iam close to get it but i miss something for shure.

Thanks for helping

CountDeMonet, 2020-02-27:
Your code is a little over complex and is missing the default states. 

`
void setCyclotronLightState(int startLed, int endLed, int state) {
  switch (state) {
    case 0: // set all leds to red
      for (int i = startLed; i <= endLed; i++) {
        if (bosonmode == true) {
          powerStick.setPixelColor(i, powerStick.Color(255, 0, 0));
        }else if (slimemode == true) {
          powerStick.setPixelColor(i, powerStick.Color(0, 255, 0));
        }else{
          powerStick.setPixelColor(i, powerStick.Color(255, 0, 0));
        }
      }
      break;
    case 1: // set all leds to orange
      for (int i = startLed; i <= endLed; i++) {
        powerStick.setPixelColor(i, powerStick.Color(255, 90, 0));
      }
      break;
    case 2: // set all leds off
      for (int i = startLed; i <= endLed; i++) {
        powerStick.setPixelColor(i, 0);
      }
      break;
    case 3: // fade all leds from red
      for (int i = startLed; i <= endLed; i++) {
        if (cyclotronRunningFadeOut >= 0) {
          if (bosonmode == true) {
            powerStick.setPixelColor(i, 255 * cyclotronRunningFadeOut / 255, 0, 0);
          } else if (slimemode == true) {
            powerStick.setPixelColor(i, 255 * cyclotronRunningFadeOut / 0, 255, 0);
          } else{
            powerStick.setPixelColor(i, 255 * cyclotronRunningFadeOut / 255, 0, 0);
          }
          cyclotronRunningFadeOut--;
        } else {
          powerStick.setPixelColor(i, 0);
        }
      }
      break;
    case 4: // fade all leds to red
      for (int i = startLed; i <= endLed; i++) {
        if (cyclotronRunningFadeIn < 255) {
          if (bosonmode == true) {
            powerStick.setPixelColor(i, 255 * cyclotronRunningFadeIn / 255, 0, 0);
          }else if (slimemode == true) {
            powerStick.setPixelColor(i, 255 * cyclotronRunningFadeIn / 255, 0, 0);
          }else{
            powerStick.setPixelColor(i, 255 * cyclotronRunningFadeIn / 255, 0, 0);
          }
          cyclotronRunningFadeIn++;
        } else {
          powerStick.setPixelColor(i, powerStick.Color(255, 0, 0));
        }
      }
      break;
  }
}
`

xblix, 2020-02-27:
No, cleaned text its block too.
Its look like fade effect its linked to something else.
iam thinking with a simple if or if else like you do i get it.
But something other block.

acfq48, 2020-02-27:
I've been working on this very project for a few months now on and off.
Got side tracked for a bit but I'm getting back on track.  I was able to
make all 4 modes with secondary fire modes for each weapon.  I've got to
debug the code to get out an issue that I've been having with boot up but I
should have that fixed within a few weeks if you care to wait.  It does run
on a Mega but might run just fine on a nano.  I haven't tested that out
yet.  If you want to wait I can send it to you when I finish.  Count, if
you want to help me debug I can send it to you in the meantime.

xblix, 2020-02-27:
> I've been working on this very project for a few months now on and off. Got side tracked for a bit but I'm getting back on track. I was able to make all 4 modes with secondary fire modes for each [...]

Cool. Same 4 mode work with secondary fire for me too and boot works perfect as iam stock for fadegame effect. But normal boot work in for mode for me.
If you have time share your code i can try debug for you too. :)

LastrisOrt, 2020-09-30:
Hey guys, what's the state here?
I'm currently assembling my proton pack. Of course, if it's possible, I would like to add the video game mode as well. What would be really cool is another (hidden) switch to toggle between movie  Cyclotron (without fading) and video game Cyclotron (with fading and four color modes).
Any ideas how we get this working?

CountDeMonet, 2020-09-30:
If you are using an arduino mega with tons of pins you could just add a switch to the pack somewhere. I use these 

https://www.amazon.com/gp/product/B01N2U8PK0/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Just read the pin high or low in the main loop then set the value of the setting bools. Then you have the ability to toggle on and off. Should even work as the pack is running. 

const bool useGameCyclotronEffect = true;   // set this to true to get the fading previous cyclotron light in the idle sequence
const bool useCyclotronFadeInEffect = false; // Instead of the yellow alternate flashing on boot/vent this fades the cyclotron in from off to red
const bool useDialogTracks = true;          // set to true if you want the dialog tracks to play after firing for 5 seconds

acfq48, 2020-10-01:
I've almost finished a large update on the code.  It adds all four video
game modes along with a secret menu for selecting different animation
speeds, colors, sounds etc.  I recently got a new job but it's time for me
to dive back into the code.  It should work on a regular Count DeMonet
setup with maybe just one more button.  You do need an arduino Mega though.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

LastrisOrt, 2020-10-01:
That's Great news. Looking forward to test your code. :-)

acfq48, 2020-10-02:
> Alex, just to confirm: I need the Arduino Mega because of the 256 kb
> flash, correct? The Arduino Nano only has 32 kb. Or is it about the pins?

The memory is important.  My code is ok but it isn't the most optimized out
there.  Hopefully, I'll get this version done before the end of the year
but having a Mega certainly doesn't hurt.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

CountDeMonet, 2020-10-03:
The nano is close to being out of pins and memory on the latest code. I had to go to lengths to get the memory usage down. If I was starting this project now, no question, I would use a mega. The only reason I went with the nano to begin with was because the original plan was to put it in the wand. When it turned out that the serial communication to the audio board and pack lights was scrambled due to noise I had to move the arduino to the pack. I had a really nice nano daughter board developed and ready for wand too.

closed 2020-11-15 by CountDeMonet