CountDeMonet/ArduinoProtonPack #15: queue array not working

JOBLOW20, 2020-03-15:
I've been using the code ''neutrino wand with graph''  and so far everything is working great.  The only thing i'm having a problem with is the queue array. For example in the code if i replace the '' playDialogTrack(playing); ''on line 397 with ''playAudio (boxTrack, playing);'' it will then play the boxtrack normally when firing between 5 to 10 seconds . What could be causing the queueArray to not work?  I am using a mega 2560 R3 and the adafruit soundboard.

CountDeMonet, 2020-03-15:
What is the problem you are having when playDialogTrack is called? Is nothing happening or are you getting a random sound?

JOBLOW20, 2020-03-15:
When the lights start flashing faster(between 5-10 seconds of holding the fire button) i let go of the firing button and the lights pause for a second and then the startup sequence happens (lights and sound).

CountDeMonet, 2020-03-15:
That is really strange. It's clearly resetting the arduino. Are you using the version of the library I have here on github or the one available in the github libraries?

https://github.com/CountDeMonet/ArduinoProtonPack/tree/master/Arduino/libraries/QueueArray

Generally when there is a reset I would look at input voltages but in this case there is just nothing that dequeueing a string would do to stress the hardware. Plus the mega has way more memory and power than the nano.

JOBLOW20, 2020-03-17:
I dont know why, but i took for granted that my source of 5 volts was ok. Turns out my power supply was to low for my board. The low limit is 6 volts. So thanks for the hint. Pretty sure its going to workout now with the right voltage supply. Thanks for the support, its really appreciated.

JOBLOW20, 2020-03-19:
The voltage was not the problem, i am going to uninstall my Arduino software and reinstall with the right libraries...

JOBLOW20, 2020-03-19:
Uninstalled my arduino software, reinstalled it and loaded the right libraries and it is still doing the same thing...  I should be receiving my nano board soon and will try with it... I'm wondering if the preferences file is causing my problem...

acfq48, 2020-03-19:
Also, double check which arduino version you're using.  I've had issues
with boards that required me to open an older version of the software to
upload my code.  It looks like I've saved version 1.6.3 and 1.8.4 for use
with my proton pack.

See if that works for you.

xblix, 2020-03-19:
can you post all your code that help to see

JOBLOW20, 2020-03-19:
I am using the 1.8.12 version and here is the code used:

#include <QueueArray.h>

#include <Wire.h> // Include the I2C library (required)
#include <SparkFunSX1509.h> // Include SX1509 library

// for the sound board
#include <SoftwareSerial.h>
#include "Adafruit_Soundboard.h"

#include <Adafruit_NeoPixel.h>

// for led triggers
#define HIGH 0x1
#define LOW  0x0

// SX1509 I2C address (set by ADDR1 and ADDR0 (00 by default):
const byte SX1509_ADDRESS = 0x3E;  // SX1509 I2C address
SX1509 io; // Create an SX1509 object to be used throughout

// bargraph helper variables
int seq_1_current = 0;  // current led in sequence 1
const int num_led = 15; // total number of leds in bar graph

// SX1509 pin definitions for the leds on the graph:
const byte SX1509_BAR_01 = 0;
const byte SX1509_BAR_02 = 1;
const byte SX1509_BAR_03 = 2;
const byte SX1509_BAR_04 = 3;
const byte SX1509_BAR_05 = 4;
const byte SX1509_BAR_06 = 5;
const byte SX1509_BAR_07 = 6;
const byte SX1509_BAR_08 = 7;
const byte SX1509_BAR_09 = 8;
const byte SX1509_BAR_10 = 9;
const byte SX1509_BAR_11 = 10;
const byte SX1509_BAR_12 = 11;
const byte SX1509_BAR_13 = 12;
const byte SX1509_BAR_14 = 13;
const byte SX1509_BAR_15 = 14;

// neopixel pins / setup
#define NEO_POWER 2 // for cyclotron and powercell
Adafruit_NeoPixel powerStick = Adafruit_NeoPixel(48, NEO_POWER, NEO_GRB + NEO_KHZ800);

#define NEO_NOSE 3 // for nose of wand
Adafruit_NeoPixel noseJewel = Adafruit_NeoPixel(7, NEO_NOSE, NEO_GRB + NEO_KHZ800);

#define NEO_WAND 4 // for nose of wand
Adafruit_NeoPixel wandLights = Adafruit_NeoPixel(4, NEO_WAND, NEO_GRB + NEO_KHZ800);

// LED indexes into the neopixel powerstick chain for the cyclotron. Each stick has 8 neopixels for a total of
// 16 with an index starting at 0. These offsets are because my powercell window only shows 13 leds. If you can show more 
// change the offset index and powercell count to get more or less lit.
const int powercellLedCount = 14;                                         // total number of led's in the animation
const int powercellIndexOffset = 1;                                       // first led offset into the led chain for the animation

// These are the indexes for the led's on the chain. Each jewel has 7 LEDs. If you are using a single neopixel or
// some other neopixel configuration you will need to update these indexes to match where things are in the chain
int c1Start = 16;
int c1End = 22;
int c2Start = 23;
int c2End = 29;
int c3Start = 30;
int c3End = 36;
int c4Start = 37;
int c4End = 43;
int ventStart = 44;
int ventEnd = 47;
[... 1066 lines cut ...]
      io.digitalWrite(SX1509_BAR_08, state);
      break;
    case 9:
      io.digitalWrite(SX1509_BAR_09, state);
      break;
    case 10:
      io.digitalWrite(SX1509_BAR_10, state);
      break;
    case 11:
      io.digitalWrite(SX1509_BAR_11, state);
      break;
    case 12:
      io.digitalWrite(SX1509_BAR_12, state);
      break;
    case 13:
      io.digitalWrite(SX1509_BAR_13, state);
      break;
    case 14:
      io.digitalWrite(SX1509_BAR_14, state);
      break;
    case 15:
      io.digitalWrite(SX1509_BAR_15, state);
      break;
  }
}

xblix, 2020-03-20:
try changing this

const bool useDialogTracks = true;

to this

const bool useDialogTracks = false;

and see if its work

JOBLOW20, 2020-03-20:
By changing it to false  the system doesnt reboot anymore,  the lights are doing what they are supposed to do but the Dialog tracks are not playing. So it is an improvement😁

xblix, 2020-03-20:
try change this part like this

// see if we've been firing long enough to get the dialog or vent sounds
      unsigned long diff = (unsigned long)(millis() - firingStateMillis);

      if ( diff > firingWarnWaitTime) { // if we are past the warning let's vent the pack
        playAudio(ventTrack, playing);
        venting = true;
        clearPowerStrip(); // play the boot animation on the powercell
      } else if ( diff > firingWarmWaitTime) { // if in the dialog time play the dialog in sequence
        if( useDialogTracks == true ){
         playDialogTrack(playing);
       }else{
          playDialogTrack(playing);
          }
      } else {
        playAudio(endTrack, playing);
      }
    }
  }

JOBLOW20, 2020-03-20:
Weird... Now its doing the same thing it did before, its rebooting.

JOBLOW20, 2020-03-20:
It Works! It was the software version. I downgraded to version 1.8.4 and now everything Works fine. Thanks for the help guys!

CountDeMonet, 2020-03-20:
interesting, thanks for the info. First I've heard of issues with the compiler software version. No idea why that would matter. I'm on 1.8.12 for the record

acfq48, 2020-03-20:
Sorry boys... but...

3 POINTS FROM DOWN TOWN!!!

Looks like Alex is coming away with the W this time.  😀

Congrats on getting it working.  If you want to try to mod the ghost trap
next check out my youtube channel. I can send you the code if you'd like.

https://youtu.be/7l_iawNhA9I

closed 2020-11-15 by CountDeMonet