CountDeMonet/ArduinoProtonPack #17: Outages and resets

LastrisOrt, 2020-09-29:
Hi,
I've just begun building the proton pack using your arduino code. When I push the fire-button, all lights are stopping for about a second before moving on. Same goes when releasing the fire-button or when reaching the 10 second limit. When it "overheats" it shuts itself down and boots again (can't tell you about the sound at the moment, I'm troubleshooting the lights first).
Things I've done so far:
- Used three different arduinos (Nano Clone, Nano Clone, Arduino Uno)
- Checked the wiring
- Changed the power supply (but always only using 5V from USB, whether it's from my MacBook or from power outlet)
- Everything powered by the USB-plug OR the 5V Vin (tested both).
- Used different versions of Arduino software 1.8.13, 1.8.12, 1.8.4 (it solved a similar problem in another thread), 1.8.5.

I hope to get some help here because I don't know what else it might be.
Thank you in advance

CountDeMonet, 2020-09-29:
If you don't have the audio board connected then you need to comment out some code cause the audio board code is blocking. 

1. comment out the contents of the playAudio function but leave the function declaration. 
2. in the main loop remove the read of the act pin.
// find out of the audio board is playing audio
  int playing = 1; //digitalRead(ACT);

I think that is enough to stop the lagging caused by the blocking code in the library. Once the audio board is connected and the files exist on it all will be well

LastrisOrt, 2020-09-29:
Awesome, thank you very much for your quick help! It worked! :-)

LastrisOrt, 2020-09-30:
It's all about the complete setup ;-)

closed 2020-09-30 by LastrisOrt