CountDeMonet/ArduinoProtonPack #19: Uploading the code

bunionhead, 2020-10-19:
Hi, is there anyone that can help me with a step by step guide and how to upload everything to the Arduino please?

Want to do the wand with graph and ecig plus add the sparks fun board.
Bit of a noob when it comes to Arduino but I’m good and following instructions 
Thanks

acfq48, 2020-10-25:
I'm always willing to help of you have any questions.  You'll want to get
all of the materials on the list first.

pe547c, 2020-11-14:
I just finished buying all of the electronics and I am working on getting the code to operate properly.

I am getting an error "QuerryArray" was not declared on line 113 of the Wand with no Bar Graph code.  There are a bunch of warnings above that but the QuerryArray shut the code down.

I did a straight Copy/Paste from the file into the Arduino IDE 1.8.13.  

Any help would be appreciated, i have not yet used Arduino.  I do have MatLab experience from college but that is about it.

Thanks

CountDeMonet, 2020-11-14:
you are missing the libraries needed for the code to work. Here is some information on Libraries and where they are installed

https://www.arduino.cc/en/Guide/Libraries

Look at the manual installation area. These are the libraries you need to put in the correct location

https://github.com/CountDeMonet/ArduinoProtonPack/tree/master/Arduino/libraries

pe547c, 2020-11-14:
In the Libraries Folder I don't see a download option in order to get the .zip file.

CountDeMonet, 2020-11-14:
On the code page there is a download repository button. Download the whole repository not just individual files. Then your life will be much easier

pe547c, 2020-11-14:
That worked like a charm :)

closed 2020-11-15 by CountDeMonet