CountDeMonet/ArduinoProtonPack #20: Proton Pack Circuit Diagram - No Bargraph Schematic

pe547c, 2020-11-15:
In the schematic I see 4 additional LEDs on the Wand and 4 additional LEDs on the Proton Pack.  Does the code cover for these and if so where do you put them?

I noticed on the video there was a blue light in the wand as well as button lights are those these LEDs?

Separate question, on the Schematic the Amplifier looks different than the one on the order sheet.  Should I be using 2 of the ones from the order sheet or could you link the model number for the one from the schematic?

Thanks a ton for this site, i am new to Arduino and this is a crash course per say :)

CountDeMonet, 2020-11-15:
In addition to the neopixel jewel, the wand has the sloblo led, top indicator led, base box front led, and vent light led. On mine I used the single neopixels for them. There are mounts for these led's in the 3d files. 
I used several amps while designing this thing. TBH I would recommend finding something more powerful if you like volume out of it. The amps/speakers can be hooked up separately as they are their own circuit so you can substitute whatever you want. This was the amp I ultimately went with 

https://www.amazon.com/gp/product/B00N42NP6W

CountDeMonet, 2020-11-15:
In the pack the additional led's are for the venting. If you have not seen them yet I have videos from the development of the electronics in a playlist here

https://www.youtube.com/playlist?list=PLBhSpY0c5z2dxnJxVPDHdh-QpJ_xY3YSR

CountDeMonet, 2020-11-15:
You can also find a ton of info in the build thread 

https://www.gbfans.com/forum/viewtopic.php?f=2&t=43273

pe547c, 2020-11-23:
I found it, all is soldered and in place now :)  thanks.

closed 2020-11-23 by pe547c