CountDeMonet/ArduinoProtonPack #21: Getting Arduino to execute

pe547c, 2020-11-23:
So i have everything wired up from the 'No Bargraph" wiring diagram.

I checked with Multimeter and every V is 4.9 to 5.1 Volts and every Data Input is 0.7V.

Toggle Switches are either 0V or .6V depending on the switch position.

I loaded the Code to the Arduino and it loaded successfully.

Problem is when i put power to the system, and flip any/all toggle switches nothing happens.  I rechecked all the Voltages and it is still as i stated above.  Arduino loads fine so i don't think that is broken.  Not sure what is wrong.

Any help would be very welcome :)

Thanks

unrepentant64, 2021-01-23:
I have a similar issue with another proton pack build that I found on Github.  Did you get any input, or solution to your project?  I replicated this build as well (as an alternative) and most of it works...

CountDeMonet, 2021-01-25:
I think discussion on pe547's issues moved over to the build thread on gbfans

https://www.gbfans.com/forum/viewtopic.php?f=2&t=43273

There is a lot of information there and the last few pages are mainly helping other people with the build