CountDeMonet/ArduinoProtonPack #22: Changing the wand LED's

bunionhead, 2020-12-16:
Hi guys, I've uploaded the code and everything is running great.
I'm just wondering if it's possible to change the 4 LED's in the wand in what they do? say for Example I'd like to make the white one flash? I've looked through the code and there seems to be a part where it mentions a white flashing LED but don't know what to do or Is this for a 5th LED that needs to be added?
Thanks guys

CountDeMonet, 2021-01-25:
Sorry I'm just seeing this. The wand chain is pretty simple to add lights to if you want. The wand lights are in their own chain so you would just add a light onto the chain and put it in the wand wherever you want.  On this line in the code update the number 4 to the number of led's in the chain. If you just add one make it a 5. 

Adafruit_NeoPixel wandLights = Adafruit_NeoPixel(4, NEO_WAND, NEO_GRB + NEO_KHZ800);

The lights are referenced in the order they are in the chain. For example the slo blo light is number 3 in the chain so to reference it and make it turn on you call the wand light helper function

setWandLightState(3, 0, 0); //set sloblow red

That function has a number of options for setting lights to specific colors and flashing and stuff 

void setWandLightState(int lednum, int state, unsigned long currentMillis){

first param is the index of the light you want to control. This is indexed 0 so the first led is 0. The state can be

0: // set led red
1: // set led white
2: // set led orange
3: // set led blue
4: // set led off
5: // fast white flashing
6: // slower orange flashing  
7: // medium red flashing
8: // fast red flashing

The last parameter is used for the flashing options. The setWandLight is only used in the main loop so if you are doing a flashing option you just include the existing param currentMillis. There are lots of examples in the code to show how to use it.

bunionhead, 2021-01-31:
Thanks for the reply. Maybe I should have explained I’m a bit of a noob at coding.
I understand the bit where you change the number to add more leds and setting the state in way(i think) but it’s the getting them to flash.

Basically I want to add 2 more leds that are white that flash alternatively