CountDeMonet/ArduinoProtonPack #24: Wand board with bar graph

dougbevan, 2021-12-17:
Great project, in many ways better than the pre-made kits, and tweak-able!

Did you happen to do a wand io board that includes the bar graph?  I'm putting together the whole project including bar graph, and it would be great to have a board that includes it.

CountDeMonet, 2021-12-17:
Glad you are finding it useful. I never did get around to ordering the boards and finishing that. I ran out of time and ended up wiring them up manually with proto boards. It would be a fun project to design that but I didn't have time. 

These are the only images I have of the bargraph proto board I built. Worked well. 
[image]
[image]

dougbevan, 2021-12-17:
Appreciate the images and response.  Yes, it's a great project and a fantastic contribution to the community.

I'm fine with programming but have never tried my hand at Eagle.  I did order the boards you designed to give them a try, but I will try extending the wand.io board to include signals/power for the bar graph.

closed 2021-12-17 by dougbevan