CountDeMonet/ArduinoProtonPack #26: Use of another soundboard

DTMC78, 2022-04-10:
First of all … thank you for your awesome work! 

As the Adafruit sound Fx is running into some limitations (only 16mb) would it be possible to switch to another soundboard without changing too much code? 

https://www.adafruit.com/product/3436

I am not really into coding, so any help would be greatly appreciated.

Dirk

CountDeMonet, 2022-04-10:
Most of the sound clips are quite small. I actually use the same board with my droid which has a lot more audio than this one. That project has about 100 audio clips and music on it you just have to be careful with how you encode the audio. 

That said, sure you could try another audio board. Make sure it is something that has an interface that the arduino can control and then you would need to modify the initialize and playback functions. Maybe have to change the way the audio is triggered but it's doable.

DTMC78, 2022-04-14:
Thanks for your reply!

Bought some SoundFX now, only issue I have is that its not recognized as an usb drive :(

And as I didnt look closely, I did not buy an [Adafruit](https://www.adafruit.com/product/2220) but instead ordered an https://www.adafruit.com/product/2341 which is the MINI version.
Any chance I will get it to work?

acfq48, 2022-10-11:
If you don't know how to code and don't want to switch sound boards, you
can instead do what I did and redo the audio at a lower quality.  The
sounds won't sound cd quality but you can fit more into the 16mb.  The
benefit of the sound board over other SD card options is in latency.  The
board trigger sounds almost instantaneously.  Some SD card players don't.

You can use audacity on PC to downstate the audio for free.  It's just a
little time consuming.  Just make sure that the audio files are the same
type as the ones that the adafruit board can accept.

Hope that helps.