CountDeMonet/ArduinoProtonPack #27: Slow bar graph

jack-626, 2022-04-28:
Hi, 

I have hooked up my bar graph on a breadboard to test it and its very slow. if i move around the sparkfun board or the breadboard it can stop or speed up to the normal speed. im not sure why this could be happening?

heres a video of it happening: https://www.youtube.com/watch?v=_VRVlzpF2B4
im still waiting for the 5 led bar graph to be delivered which is why im using 2x 10 led ones
and i couldn't find any resistor arrays in stock so im using normal 330 ohm resistors. 
could these changes cause the issues?

Thanks!

CountDeMonet, 2022-04-28:
do you have the soundboard hooked up?

jack-626, 2022-04-28:
Thanks for the quick reply!
I just tried it with the soundboard plugged in and its running at the right speed now!
Thank you for the help!

closed 2022-04-28 by jack-626