CountDeMonet/ArduinoProtonPack #30: Powercell not working properly

ghostbuster84, 2023-10-11:
While powered on, the powercell boots up with multiple different colors on the LEDs, and only turns on 12 lights. [image]

Here is a picture for reference when fully on. Is this normal for the spirit code, or is something wrong with my LEDs?

acfq48, 2023-10-13:
> the LEDs, and only turns on 12 lights. [image:
> Screenshot_20231011_173001_Studio.jpg]

That doesn't look right to me.  It should all be blue on the power cell
lights.  It does look like the animation sequence is correct.  Make sure
that you have true RGB neopixels and not just a strip with only certain
colored pixels.

ghostbuster84, 2023-10-13:
So, does the animation only do 12 of the 16 lights? Also, the sticks say they are rgbw, so I'd think it should be working. I guess I might have to order another set to make sure.

CountDeMonet, 2023-10-13:
There are different multi led sticks that are triggered differently. I've run into this once before. Do you have a link to the sticks you purchased? The detail you are looking for in the specs for the sticks are "WS2812 5050 RGB LED"

You can increase the number of led's used in the animation. On my spirit pack I could only fit 12 of the led's in the available area so I made it configurable. Look for the line that says

const int powercellLedCount = 14;   // total number of led's in the animation

ghostbuster84, 2023-10-13:
This is the one i bought the first time: Adafruit Industries NeoPixel... https://www.amazon.com/dp/B08Y65TQXL?ref=ppx_pop_mob_ap_share

ghostbuster84, 2023-10-13:
I just bought another set, but using the link you have in the Readme file. I guess I didn't do that for that last one.

CountDeMonet, 2023-10-13:
this is the correct one for future reference. 

https://www.amazon.com/Adafruit-NeoPixel-Arduino-Integrated-Drivers/dp/B00IEDH26K/?

you can probably get those cool white ones to work but you would need to play with the code to get the correct colors. I've never worked with the cool white ones before and it appears they are either setup differently or the colors are set differently.