CountDeMonet/ArduinoProtonPack #32: Animation is very slow

NvdGoorbergh, 2024-09-19:
Hi, I am trying to make a 3d printed proton pack and came across your github page. I had an arduino laying around with some jewels and bars and thought to give it a try. When I loaded the code onto my Uno or Nano the animation was playing correctly but was very slow. So slow that the startup animation took almost 2 minutes to finish. 

It also seems to skip a beat after a few steps.
Is that something you also recognize? It happens on both micro controllers.

CountDeMonet, 2024-09-19:
which code are you using? 

I have seen slowness if the audio board is not connected or not setup correctly.

NvdGoorbergh, 2024-09-19:
That is a good one. I was planning on adding sound later so I used the non spirit code. 
will check out the other code and see if that works. (After receiving new parts, accidentally burned trough my led bar 😅)

thanks! Will let you know if that solved it :).

CountDeMonet, 2024-09-20:
You can comment out the code that calls the audio board. The slowdown is
caused by something in the library for the board.

The spirit code is wired slightly differently from the full pack because I
wanted to make it easy to setup. Just be aware of that

NvdGoorbergh, 2024-09-21:
Thanks! Will chexk it out when the replacement parts arrive. How many wires are going trough the loom to the wand? I would like to have the quickconnect but need to know how many pins are needed 😅